OrcaSlicer/OrcaSlicer #10734: 3MF file import shuts down orca 2.3.1 beta

TAGNG, 2025-09-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1 Beta

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro build 26100

### Additional system information

8th Gen Core i7, 32Gb ram, 8 TB storage, running Windows 11 Pro Build 26100

### Printer

Qidi MAX 3

### How to reproduce

1. Import the saved 3MF file or project 
2. Orca 2.3.1 crashes and shuts down

### Actual results

Orca Shutdown and you have to relaunch

### Expected results

import 3MF files for slicing

### Project file & Debug log uploads

Nothing to upload orca shuts down

### Checklist of files to include

- [ ] Log file
- [ ] Project file

armaguedes, 2025-09-16:
I just downloaded the Portable version, and launched it (while having 2 instances of installed OS 2.3.0 launched). Imported some 3MF I have; it loaded fine, and also sliced just fine.

TAGNG, 2025-09-16:
The issue is not with 2.3.0 it is with 2.3.1 betaRegardsEugene

TAGNG, 2025-09-16:
The 3mf files were saved on 3.2.1 alpha. It works fine with step and stl files, but Falls over the moment we try importing  3.2.1 alpha createdfilesRegardsEugene