OrcaSlicer/OrcaSlicer #10752: Bed Layout

KennyGreg, 2025-09-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-beta

### Operating System (OS)

Windows

### OS Version

Windows 11 pro version 24H2

### Additional system information

AMD Ryzen 7 3700X 8-Core Processor              (3.60 GHz)
32.0 GB ram
NVIDIA GeForce RTX 2060 SUPER

### Printer

Bambu Labs P1S

### How to reproduce

Open Orca Slicer,
Go to Prepare
The bed is out of align with its demensional markings in the preview window
See image

[image]

### Actual results

The bed image does not match the demension image.

[image]

### Expected results

The bed and its demesions should line up

[image]

### Project file & Debug log uploads

No project started.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2025-09-16:
Confirming this behavior when opening with **Orca Slicer** a **`.3mf`** saved with _BaumbuStudio_.

yw4z, 2025-09-17:
as you can see a model file set in project. orca also imports that value (but souldnt)

[image]

but value not visible in dialog

[image]