OrcaSlicer/OrcaSlicer #10774: lineas intercaladas en el rellono panal de abeja

Sircriatura, 2025-09-17:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

no puedo desactivar la linea intercalada en el relleno de panal de abeja, no me gusta como queda y no puedo desactivarlo

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

quisiera poder desactivar esa nueva funcion

ianalexis, 2025-09-18:
A que te referís con lineas intercaladas?
Si te referís a como el panal de abeja hace paredes con doble recorrido en 2 de 6 paredes de cada hexagono:
[image]
Es algo ventajoso ya que optimiza el recorrido enormemente y al mismo tiempo mejora su fuerza al depositar una linea sobre la union de 2 de la capa anterior.

Si te referís a otra cosa por favor dame algún mejor ejemplo.

---

What do you mean by intercalated lines?
If you are referring to how honeycomb makes double-layered walls on 2 of the 6 walls of each hexagon.
This is advantageous because it greatly optimizes the layout and at the same time improves its strength by depositing a line on the junction of 2 of the previous layer.
If you are referring to something else, please give me a better example.

Sircriatura, 2025-09-25:
me refiero exactamente a eso, solo quiero que haga una linea como siempre, todo lo que lamino ahora no hay manera de que salga bien, las litofanias tambien me salen mal, no se que ha pasado pero no funciona bien el laminador

ianalexis, 2025-09-26:
No entiendo lo que decís.

> que haga una linea como siempre

¿Qué es “como siempre”? Si nunca funcionó así.
No existe una manera óptima de generar un honeycomb con una sola línea: en el mejor de los casos terminarías con hexágonos individuales que crean paredes dobles (más uniformes, pero dobles al fin). La otra alternativa serían líneas únicas con múltiples retracciones, recorridos sobre paredes ya impresas y un tiempo de impresión mucho mayor.

Quizás deberías considerar otro patrón de relleno. En la [Wiki de patrones](https://github.com/SoftFever/OrcaSlicer/wiki/strength_settings_patterns#patterns-quick-reference) están explicados con sus ventajas y desventajas.

Si lo que buscás es algo similar pero evitando esas “dobles paredes”, podrías probar con:

- [3D Honeycomb](https://github.com/SoftFever/OrcaSlicer/wiki/strength_settings_patterns#3d-honeycomb)
- [Cross Hatch](https://github.com/SoftFever/OrcaSlicer/wiki/strength_settings_patterns#cross-hatch)
- [TPMS-D](https://github.com/SoftFever/OrcaSlicer/wiki/strength_settings_patterns#tpms-d)

> las litofanias tambien me salen mal

¿Cómo, y qué relación tiene con este issue? ¿Podrías subir fotos y el proyecto que “sale mal”?