OrcaSlicer/OrcaSlicer #10777: 2.3.1-beta - Clone quantity not updating properly

ThepHuck, 2025-09-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-beta

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro 24H2

### Additional system information

CPU: Intel i5-13400F
64GB of RAM
NVidia RTX 4060Ti

### Printer

Elegoo Centauri Carbon

### How to reproduce

Add an object to the build plate
Select the object, right-click, select Clone, enter a number, such as 19.  Result is 20 objects automatically placed on the build plate
CTRL-Z to remove the 19 objects and revert back to single object
right-click on the object, select clone, leave the number at 1.  Result is 19 additional items.
CTRL-Z to remove the 19 objects and revert back to single object
right-click on the object, select clone, change number to 2.  Result is still 19 additional items.
If you delete all but one item, then right-click and clone, result is the desired quantity.
It only happens when CTRL-Z is used to revert the clone.

### Actual results

Quantity of objects cloned stays at 19.

### Expected results

Clone the items to the number specified.

### Project file & Debug log uploads

[debug_Mon_Sep_15_16_28_05_3168.log.zip](https://github.com/user-attachments/files/22395573/debug_Mon_Sep_15_16_28_05_3168.log.zip)

[mushroom-dots.zip](https://github.com/user-attachments/files/22395577/mushroom-dots.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

MxBrnr, 2025-09-18:
This sounds like normal behavior to me because the number you enter into the clone box is the number of clones you want to _create_, not the total number of items on the plate.

ThepHuck, 2025-09-18:
Y’all are misunderstanding the issue.  The issue is that after the CTRL-Z, the number you enter into the clone box is ignored and it uses the previous number you entered.

For instance, I used 19 because that’s the number I used, which resulted in 20 objects.  I hit CTRL-Z to revert to only 1 object on the plate, tried to clone 1, which should result in 2 objects, but instead it created 19 clones.  I hit CTRL-Z again, which removes the 19 and reverts to 1.  I selected clone and entered 4, it still created 19 clones, resulting in 20 total objects again.