OrcaSlicer/OrcaSlicer #10794: Gap between infill or part and infill of modifier

J12345T, 2025-09-19:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1

### Operating System (OS)

Windows

### OS Version

Win10

### Printer

Bambulab x1c

### How to reproduce

1. load a part into the slicer
2. set 1 wall and 2% infill
3. load a modifier part into the slicer
4. set 1 wall and 10% infill
5. 0verlap the part and the modifier
6. slice the part
7. note the small gap between the infill of the part and the infill of the modifier 

### Actual results

The result is that the 2 infills do not bond together because of the aforementioned gap. 

### Expected results

The expectation is that the 2 infills (part and modifier) should have no gap between them so that they bond together in the print. 

### Project file & Debug log uploads

[Gap Issue.zip](https://github.com/user-attachments/files/22419421/Gap.Issue.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

Screenshots included in the zip file. Thanks.

[image]
[image]
[image]

RF47, 2025-10-07:
you can try add 1mm anchor length, meanwile.

RF47, 2025-10-07:
> Thank you. I tried that and it did nothing to adhere the two infills. It just made the gap cleaner.
> 
> FWIW both Cura and Prusia slicers slice this model correctly.

As far as I know, neither Prusa Slicer nor Cura have lateral lattice infill.

J12345T, 2025-10-07:
Yes, you're absolutely correct which is why I need Orca devs to fix this issue..