OrcaSlicer/OrcaSlicer #10815: Missing graphical preview on Ubuntu

JaCKoP619, 2025-09-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

v2.3.1-beta

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04

### Printer

irrelevant

### How to reproduce

Install either via appimage, or as flatpak from the official webpage
open orca and create new project
open preview tab

### Actual results

missing project graphical preview

[image]

### Expected results

Working graphical preview

### Project file & Debug log uploads

i got those errors from terminal:
./OrcaSlicer_Linux_AppImage_Ubuntu2404_V2.3.1-beta.AppImage [2025-09-20 14:15:18.513881] [0x00007dbf561484c0] [trace] Initializing StaticPrintConfigs add font of HarmonyOS_Sans_SC_Bold returns 1 add font of HarmonyOS_Sans_SC_Regular returns 1 add font of NanumGothic-Regular returns 1 add font of NanumGothic-Bold returns 1 (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.034: gtk_window_resize: assertion 'height > 0' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.046: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.046: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.047: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.047: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.048: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.048: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.048: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.048: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.048: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.049: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.049: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.049: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.049: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.050: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.215: gtk_window_resize: assertion 'height > 0' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.217: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.217: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.217: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.217: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.217: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.217: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.217: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.217: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.221: gtk_window_resize: assertion 'height > 0' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.221: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.221: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.222: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.222: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.222: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.222: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.222: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.222: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.222: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.225: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.225: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.265: gtk_window_resize: assertion 'height > 0' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.268: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.268: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.268: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.268: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.273: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.273: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.273: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.273: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.274: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.274: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.274: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.274: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.278: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.278: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.278: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.278: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.281: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.281: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.281: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.281: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.284: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.284: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.285: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.285: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.288: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.288: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.288: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.288: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.290: gtk_widget_set_size_request: assertion 'width >= -1' failed (orca-slicer:41805): Gtk-CRITICAL **: 14:15:22.290: gtk_widget_set_size_request: assertion ' [... cut ...] _IS_CELL_LAYOUT (cell_layout)' failed

Log files:

[debug_Sat_Sep_20_14_13_10_41209.log](https://github.com/user-attachments/files/22440935/debug_Sat_Sep_20_14_13_10_41209.log)

[debug_Sat_Sep_20_14_15_18_41805.log](https://github.com/user-attachments/files/22440940/debug_Sat_Sep_20_14_15_18_41805.log)

[debug_Sat_Sep_20_14_18_10_42360.log](https://github.com/user-attachments/files/22440942/debug_Sat_Sep_20_14_18_10_42360.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Khyzir, 2025-09-24:
I had a similar problem today. Below is my system info,  Issue and fix.

System Information
CPU: Intel Core i3-12100F
GPU: NVIDIA GeForce GTX 1050 Ti
Display: 1920x1080 @ 60Hz ZOWIE XL
Memory: 7.29 GiB / 31.15 GiB (23%)
OS: Arch Linux x86_64
Kernel: Linux 6.16.8-arch1-1
DE/WM: KDE Plasma 6.4.5 / KWin (Wayland)
Shell: bash 5.3.3

3D Printer
Creality Ender-3 V3 SE running Klipper on Raspberry Pi 4

Issue
Updating to OrcaSlicer-Linux-flatpak_V2.3.1-beta_x86_64.flatpak causes a blank screen when opening a project.
Downgrading to OrcaSlicer-Linux-flatpak_V2.3.1-alpha_x86_64.flatpak does not resolve the issue.

Temporary Fix

sudo flatpak uninstall org.freedesktop.Platform.GL.nvidia-580-82-09
sudo flatpak install flathub org.freedesktop.Platform.GL.nvidia-570-144

Additional Info

Host GPU driver: NVIDIA-SMI 580.82.09 (CUDA 13.0)
Problem seems related to Flatpak runtime and NVIDIA GL compatibility.

JaCKoP619, 2025-09-24:
damn, tampering with nvidia drivers usually ends up with a full linux reinstall for me...

> I had a similar problem today. Below is my system info, Issue and fix.
>
> System Information CPU: Intel Core i3-12100F GPU: NVIDIA GeForce GTX 1050 Ti Display: 1920x1080 @ 60Hz ZOWIE XL Memory: 7.29 [...]

Khyzir, 2025-09-24:
> damn, tampering with nvidia drivers usually ends up with a full linux reinstall for me...

I think OrcaSlicer Flatpak runs in a sandbox and doesn’t use your system NVIDIA drivers directly. Instead, it installs and uses Flatpak GL runtimes for GPU support.

JaCKoP619, 2025-09-25:
doubt it, i ran first appImage then flatpak, with same result

Tritbool, 2025-10-04:
Same problem happens on ubuntu 25.10 aarch64
It happens both with : 

1. The flatpak file
2. The binary I built from source on my machine

Graphics are Nvidia 580 on a RTX 3070TI. Orca slicer visible in nvidia-smi

JaCKoP619, 2025-10-05:
Same for me, switching from 580 to 575 Nvidia driver fixed it for me

JaCKoP619, 2025-10-19:
this fixed it for me as well, will do some more testing, but great catch man!