OrcaSlicer/OrcaSlicer #10818: No Flushed amount in Preview Window

Beagon, 2025-09-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-beta

### Operating System (OS)

Linux

### OS Version

Manjaro

### Printer

Creality K2 Plus

### How to reproduce

 - Make a multi colour print for a K2 Plus.
 - Slice and check Preview tab.
 - See no Flushed amount

### Actual results

[image]

### Expected results

(Taken on default Bambu A1 Profile)

[image]

### Project file & Debug log uploads

3MF of Both
[BlockC.zip](https://github.com/user-attachments/files/22447010/BlockC.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

SoftFever, 2025-09-21:
Hi there,

I don’t see the flush configured in your sample K2 project.  
In Orca, the flush action is recognized by finding G-codes wrapped with `; FLUSH_START` and `; FLUSH_END`.  
Orca calculates the flush amount accordingly and displays it in the preview window.

For example, you can find the flush custom G-code below in the Bambu or Qidi Q2 printer’s configuration.

[image]

[image]

Beagon, 2025-09-21:
> Hi there,
> 
> I don’t see the flush configured in your sample K2 project. In Orca, the flush action is recognized by finding G-codes wrapped with `; FLUSH_START` and `; FLUSH_END`. Orca calculates the flush amount accordingly and displays it in the preview window.
> 
> For example, you can find the flush custom G-code below in the Bambu or Qidi Q2 printer’s configuration.
> <img alt="Image" width="1034" height="704" src="https://private-user-images.githubusercontent.com/103989404/491997951-ac933130-e5cc-42ad-bf30-fd91ae823d6b.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg0NDUzNTEsIm5iZiI6MTc1ODQ0NTA1MSwicGF0aCI6Ii8xMDM5ODk0MDQvNDkxOTk3OTUxLWFjOTMzMTMwLWU1Y2MtNDJhZC1iZjMwLWZkOTFhZTgyM2Q2Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkyMVQwODU3MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYzI5YWFlYzMzZTYwMjlmNTY1NTY5OTZiMDNmMWJkZmJiMzkwOTAxNjNmZTVhYjNjNjYzNmRmNzU0NzY0MTViJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.n8S891Hekt_HFRAAFxt_yDUQIyH4RCiMy5XuzfnP8fk"> <img alt="Image" width="194" height="317" src="https://private-user-images.githubusercontent.com/103989404/491997864-4808dc82-4be9-46f3-bb39-52748bfd7991.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg0NDUzNTEsIm5iZiI6MTc1ODQ0NTA1MSwicGF0aCI6Ii8xMDM5ODk0MDQvNDkxOTk3ODY0LTQ4MDhkYzgyLTRiZTktNDZmMy1iYjM5LTUyNzQ4YmZkNzk5MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkyMVQwODU3MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZjRlMGVjMDA0OTdmZmI5ZTMzMjE3ZjBmMjE5YTIwZTAwZjc2ZGY1OThiNGUzMTQ4ODUzZmE0ODA5ZjIyZmZlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.oYBiLAdavEMwDqgvmUn8RaeIho1oSboAALZfzvq_M9Q">

The K2 Config is the default one that comes with Orca. I would assume it came preconfigured?

Stevetm2, 2025-11-08:
The below change describes how to add the EXTERNAL_PURGE comment to the K2 Printer's Filament Change gcode, so the flush amount can be calculated. Try it without the T command, on the K2, unless you have that  command in there already.
https://github.com/SoftFever/OrcaSlicer/pull/7508