OrcaSlicer/OrcaSlicer #10840: Support interface ironing does not generate with Tree Slim/Strong/Hybrid style supports

cheatreal321, 2025-09-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-beta

### Operating System (OS)

Windows

### OS Version

Windows 10 Enterprise LTSC

### Printer

Qidi Tech X-Max 3 / Prusa XL

### How to reproduce

1. Load any model with overhangs or floating regions that need supports
2. Enable supports and choose "Tree (auto)" with sufficient threshold angle
3. Select "Tree Slim", "Tree Strong", or "Tree Hybrid" supports
4. Enable "Ironing Support Interface"
5. Slice model to see that no ironing is happening compared to when using "Organic" style supports

### Actual results

#### Example of Tree Hybrid supports generating without any support interface ironing
[image]

### Expected results

#### Meanwhile Organic supports correctly generate with support interface ironing at the exact same layer
[image]

### Project file & Debug log uploads

[project file and debug log.zip](https://github.com/user-attachments/files/22478995/project.file.and.debug.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

On the off-chance that this is just not implemented yet rather than a bug, I would highly suggest that the "Support ironing" category disappear from the UI for the time being when a style of support that is not yet compatible is selected. Then at the very least other users won't waste their time trying to get it to work like I did.

Trasterin, 2026-05-28:
Woah, Almost 8 Months and the problem Stills in the last build.

karailker, 2026-09-21:
A fix is open in #15159.

Tree Slim, Strong and Hybrid build their toolpaths in TreeSupport::generate_toolpaths(), which never read the ironing settings. Organic goes through the shared support code, which already irons the top contact layer. The PR adds the same ironing to the normal tree path, on the roof area that touches the model.

I tried it on a 40x40x4 mm plate sitting on a 10x10x8 mm stem, with tree(auto), Tree Slim, 3 top interface layers and support ironing on. On main, layer 63 (Z=7.80) shows only the interface lines. With the PR, the same layer is ironed. The Line Type legend gains an Ironing row, and the estimated time goes from 1h36m to 1h51m.

Before (main):
[image]
[image]

After (PR):
[image]
[image]

In the G-code, main has no ;TYPE:Ironing block. With the PR there is one at Z=7.8, right after the top interface layer. Slim, Strong and Hybrid all iron the top interface layer with 1, 2 and 3 top interface layers. Organic output is unchanged, and the G-code is identical with support ironing off.

G-code before after:
[before] before_tree_slim_n3.gcode: 125654 lines
  ';TYPE:Ironing' blocks: 0
  ';TYPE:Support interface' blocks: Z=7.21538, Z=7.50769, Z=7.8
[after] after_tree_slim_n3.gcode: 127598 lines
  ';TYPE:Ironing' blocks: 1  at line 116743 (Z=7.8, 1838 extrusion moves)
  ';TYPE:Support interface' blocks: Z=7.21538, Z=7.50769, Z=7.8

[after] excerpt around the ironing block:
  G1 X78.283 Y98.387 E.01213
  G1 X78.143 Y98.552 E.00887
  G1 X78.143 Y101.684 E.12834
  M204 S10000
  G1 X79.75 Y102.349 F7200
  M204 S500
  ;TYPE:Ironing
  ;HEIGHT:0.02
  G1 F1200
  G1 X79.75 Y97.706 E.01528
  G1 X79.65 Y97.722 E.00033
  G1 X79.65 Y102.278 E.01499
  G1 X79.55 Y102.261 E.00033
  G1 X79.55 Y97.739 E.01488

This covers Slim, Strong and Hybrid. I did not test Snug, so the partial ironing @sakarnen describes there is not part of this PR.

If someone can try the PR build on the model from the report, that would help.