OrcaSlicer/OrcaSlicer #10866: Filament profile for BBL PLA Tough+

mark8055, 2025-09-25:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

It looks like BBL PLA Tough+ has some different settings from regular PLA, is it possible to have a profile for Orcaslicer? 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

A new profile for the filament

lightmaster, 2026-01-03:
Seeing as this new filament profile request has been sitting around for almost half a year, does anyone have a custom profile they can share for it?

neo-jayfeather, 2026-01-03:
This is available in the nightly build and were added before the release but not included in the release. You can manually copy the settings from a nightly build if you want to use a stable version (copy the .json files into the resources/profiles/BBL or wherever they go), or switch to a nightly build https://github.com/OrcaSlicer/OrcaSlicer/releases/tag/nightly-builds.