OrcaSlicer/OrcaSlicer #10882: Definable Variable Layer Height

jefftopp1984, 2025-09-27:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I don't really like the current variable layer height method because you can't actually set it to change exactly where you want, you can't actually get every layer to the right number with right and left clicking the mouse and you can't change the print rules.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Be able to select the layer the same way you would select a layer for a color change or any other custom Gcode and then change the print preset at that point. So, say my model is straight up and down at the beginning but has curves in the walls toward the top. I wouldn't want the whole print using the 0.1mm preset with the correct speeds for the flow rate at that layer height and then have it switch to 0.3mm using the same rules otherwise. I'd like to start the print using the 0.3mm preset and then change at a specified layer to the 0.1mm preset so that all of the speeds and other settings change along with it. The column along the side is cumbersome and doesn't allow you to change the print settings at the layer change height. The variable Layer Height needs to be able to be assigned to presets which have the speeds and other things defined in them that must change along with the layer height.

### Describe alternatives you've considered

There are none

coryrc, 2025-10-01:
There are definitely alternatives. You can measure and enter the number in the height range box.

jefftopp1984, 2025-10-02:
> There are definitely alternatives. You can measure and enter the number in the height range box.

How without taking forever and how do you switch all of the speeds and other rules that need to change with the layer height? I don't want to print 0.1mm at 0.3mm speed when I can print the 0.1mm layers at 2.5x the speed. I am volumetrically limited to 110mm/s at 0.3mm layers. I hit a frame resonance above 250mm/s but there is still flow overhead at 0.1mm. But if I have a model that is very simple at the base but becomes complex at the top, I don't want to have the 0.1mm layers printing at 110mm/s. That defeats the purpose of saving time with variable layer height. It needs to be able to switch to an entirely different print preset at a given layer.