OrcaSlicer/OrcaSlicer #10901: YOLO Flow Calibration Ignores Top Surface pattern

malinhiles, 2025-09-29:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Ender 3 V2

### How to reproduce

1. Launch OrcaSlicer v2.3.0
2. Go to "prepare" tab with a clean buildpate
3. Click on Calibration tools dropdown and select "YOLO (Recommended)" 
4. With Yolo (Recommended)" on the build plate, an visual inspection of the slice shows that the top surface pattern is a monotonic line.
5. Go to the Top Surface Pattern setting select the "Archimedean chords" and reslice.
6. Observe how the top surface is still monotonic line.
7. Go to select any other top surface pattern and reslice.
8. Observe how the top surface is still monotonic line.

### Actual results

[image]

### Expected results

Resultant top surface pattern should reflect the one selected in the slicer settings

### Project file & Debug log uploads

[Orca YOLO Flow Calibration.zip](https://github.com/user-attachments/files/22592874/Orca.YOLO.Flow.Calibration.zip)

[debug_Sun_Sep_28_12_12_21_17792.log.zip](https://github.com/user-attachments/files/22592901/debug_Sun_Sep_28_12_12_21_17792.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

malinhiles, 2025-09-29:
> You have to change it for each individual chip in the Object tab.

Copy, even when changing it via the global tab, It still needs to be specifically changed on a per object basis? Is intentional behavior of the slicer? 

> Also, using the Archimedean Chord pattern in v2.3.0 will not work the same as in v2.3.1, where the pattern has been custom programmed to print in a specific sequence in order to give the correct results.

Good to know. For my purposes of flow calibration in 2.3.0, does that mean that I should just use the old monotonic 2-pass method instead?

@MxBrnr Thank you for your help and insight

MxBrnr, 2025-10-06:
> It still needs to be specifically changed on a per object basis? Is intentional behavior of the slicer?

In this case, yes, and yes.

> does that mean that I should just use the old monotonic 2-pass method instead?

You can still use the YOLO tests. Only the pattern changed between the versions, the 4 tests (Pass 1, Pass 2, YOLO, and YOLO perfectionist) exist in both.

malinhiles, 2025-10-06:
@MxBrnr, Thanks for the insight!